cortex-lab/Rigbox #1: functionality of abort button: delete experiment entry from database, and don't log water

nsteinme, 2017-02-08:
Otherwise, log water automatically on end, and report how much more is required in obj.log.

nsteinme, 2017-02-08:
In https://github.com/cortex-lab/Rigbox/commit/b11354862bc01cd4d376f9a9304ab4c1fd8b59ed, added the water posting but not the experiment deletion. Added to the place where data is saved in exp.Experiment, but in future could be a "service"? Don't know how to use them. Currently untested.

k1o0, 2018-05-08:
Water no longer posted when the experiment is aborted (1a92b22df00cc161cfa564cae07d9de1e7757d69), however I don't think we should delete any Alyx file records.  If you agree, let's close the issue, otherwise I'll bring it up in the next meeting.

nsteinme, 2018-05-09:
don't have strong feelings - closing -

closed 2018-05-09 by nsteinme