cortex-lab/Rigbox #4: Task bar visible after gamma calibration

k1o0, 2017-02-10:
After calibrating using 'gamma calibration' branch code the plotting of figures by the function in Window causes psychtoolbox's Screen window to be behind the Windows task bar (in Windows 7, at least).  It remains this way until you manually select the window to again focus it. 

Three possible solutions:
1. have it restart expServer after calibration.
2. writing in a figure suppression flag
3. perhaps some way of telling psychtoolbox or the OS to refocus the window

nsteinme, 2017-02-10:
Do you mean that it does that every time you subsequently run expServer, even after restarting the computer? Or only right then, that particular time just after you do the calibration? If the latter I think it's probably fine to just restart after you do it, no? If the former, it sounds like a bug in your calibration code, good thing we haven't merged it with master yet ;)

P.S. if you want to draw someone's attention to a thread, but without necessarily "assigning" them, you can just use a twitter-like tag: @k1o0 . It will email them about it.

k1o0, 2017-02-10:
No it's caused by hw.ptb.Window/calibration opening figures so it would only ever occur when the calibration is run.  Manually restarting expServer or selecting the OTB window with the mouse fixes it.

k1o0, 2019-11-04:
Needs real-world testing but closing for now.

closed 2019-11-04 by k1o0