cortex-lab/Rigbox #12: condition numbers

nsteinme, 2017-03-04:
Would be great for a Parameters object to have a special field "condition number" that just counts 1:n for however many conditional parameters there are, then writes the appropriate number into block.trials.cond. 

It could also display this in the conditional panel, but it would have to be a read-only field.

nsteinme, 2017-03-04:
Useful for mimicking certain useful features of mpep: https://github.com/cortex-lab/Rigbox/issues/13

k1o0, 2018-12-12:
Although this has been a feature for years, it's undocumented.  To do:

- [ ] Document solutions to issues #13 & #12 on Wiki and instructions. 
- [ ] Add randomiseConditions parameter to default parameters or maybe create special checkbox above conditions table?

k1o0, 2019-05-02:
Wiki updated, closing this.

closed 2019-05-02 by k1o0