cortex-lab/Rigbox #15: "save parameters as next" button

nsteinme, 2017-03-18:
Just a quick idea, would be useful to have a button that saves the current parameters selection into the "last" location (i.e. most recent date, last expNum+1) for the selected mouse so they get loaded next time for that mouse, as an alternative to starting a fake block and aborting it. This would be faster and also make it clearer what data corresponds to something with a mouse, versus not.

k1o0, 2017-03-20:
Seems a little unnecessary to me.  Currently one can quickly run the session or save the profile into the saved sets.  Adding another button to do something very similar doesn't seem worth it for the length of time saved.

nsteinme, 2017-03-20:
This was not an assignment to you :) Don't implement it if you don't feel like it. The primary gain is that we don't have to worry about how to deal with those fake sessions in terms of database logging - it would be best if only real data gets logged. Currently there is no way to delete stuff from the database (not from the interface we have in matlab anyway) so if you create the entry at the beginning of the experiment, it will get saved even if you later click "abort".