cortex-lab/Rigbox #17: refactor eui.ParamEditor to modify graphical elements rather than re-create

nsteinme, 2017-03-21:
- [ ] eui.ParamEditor needs a new function, "populate", which accepts a exp.Parameters object and re-fills existing fields of the ParamEditor with these new data. Makes invisible any un-used fields. Creates extras if there were previously not enough fields.

- [ ] Ideally, the ParamEditor's initial build function would be split into one that creates all of the "base" graphical elements, like all the boxes and panels, the conditional table, and a set of maybe 20 of the global parameter controls, and then itself calls the populate method to fill this empty base. 

- [ ] Alternatively, a different kind of faster scheme could be devised. E.g. the Global panel could be replaced by a uitable (just like Conditional), and the "CellSelectionCallback" can be used to make the "conditionalize" button still work. I think you might lose the tooltips though.

jaib1, 2018-12-12:
In general, should re-do the ParamEditor, maybe from scratch, maybe based off Burgess' code in the +ui folder in signals

k1o0, 2019-04-29:
Performance has been sufficiently improved with the most recent changes to the param editor.

closed 2019-04-29 by k1o0