cortex-lab/Rigbox #24: Minimum water requirement in Alyx not consistent with google sheets

laurax452, 2017-04-25:
Water requirement in Alxy is different from the minimum water requirement currently in google sheets for most/all(?) animals. Can't see where discrepancy lies.

nsteinme, 2017-04-25:
Thanks, @k1o0 could you investigate? First check implant weight, ref weight, age, gender all match for some mouse. Then whether z-score matches, which will help determine what part of algorithm diverges.

laurax452, 2017-04-25:
Miles and I checked the things like implant weight, age, z score and gender for a couple of mice- e.g. Vin, Diesel, Kendall.
They all matched – I haven’t checked for all mice but essentially I think they all match.
Alk041 has a z score that doesn’t match, there may be more examples of this.

nsteinme, 2017-04-25:
Ok, great. This is an alyx issue so I'm going to move it there.

closed 2017-04-25 by nsteinme

k1o0, 2017-04-25:
I haven't updated the code yet so let's hold off until tomorrow