cortex-lab/Rigbox #33: register the params file to alyx

nsteinme, 2018-01-11:
in dat.newExp, I think.

k1o0, 2018-01-17:
Parameter files are currently not a valid dataset type. Do you know how to add new ones?

nsteinme, 2018-01-17:
Yes, just from the admin interface, under "data files" subsection, nothing
to it. Can I recommend that we write out a copy of the params in json and
register that file instead of the mat file? Since the parameters variable
is a struct, this should work seamlessly, and the json will be more
transparent/openly-accessible. Only thing to consider is whether we'd lose
something by not having function handles, or what happens when you try to
save a json struct with a function handle in one of the fields, then
re-load it? Could you test that, and maybe more generally just test
saving/loading a few parameters sets into and out of json to see that they
come out the same (and can still be opened with eui.ParamEditor, for
example)?

k1o0, 2018-01-24:
Params are now saved and registered to Alyx as JSON if availiable, or as a mat if not.

closed 2018-01-24 by k1o0