cortex-lab/Rigbox #37: include git commit hashes in parameters struct

nsteinme, 2018-01-26:
For timeline.hw, also for block/experiment.
How to extract them from relevant repositories?

mkrumin, 2018-01-26:
if you have git installed, and runnable from cmd/powershell

[s, hash] = system('git rev-parse HEAD')

nsteinme, 2018-01-26:
Awesome, thanks for that. Note to self or @k1o0 :
`[s, hash] = system(sprintf('git -C "%s" rev-parse HEAD', fileparts(which('funcName'))))
`
where funcName is a function in the root of the repository, like "mc" for Rigbox, or "addSignalsPaths" for signals.

k1o0, 2018-02-13:
The parameters struct is saved by mc, not expServer, and we want to know what code the stimulus computer is running.  Loading the file and re-saving is not ideal...

nsteinme, 2018-02-13:
Good point - we don't really want it with the parameters struct anyway - we want it with the hardware info: #40 

My mistake, but I still think it's a good idea to have in this other location.

k1o0, 2018-05-08:
a27c33aeb488f58dabd8a4a7897e61802a51ac28

closed 2018-05-08 by k1o0