cortex-lab/Rigbox #51: Optional ext in expFilePath

k1o0, 2018-05-09:
Should have an optional input defining the file extension of the resulting path, with the default in most cases being 'mat':
```
% [full, filename] = expFilePath(ref, type[, reposlocation, ext])
[...]
  function [repos, suff, ext, dateLevel] = typeInfo(type)
    repos = 'main';
    switch lower(type)
      case 'block' % MAT-file with info about each set of trials
        suff = '_Block';
        ext = 'mat';
      case 'hw-info' % MAT-file with info about the hardware used for an experiment
        suff = '_hardwareInfo';
        ext = 'mat';
      case '2p-raw' % TIFF with 2-photon raw fluorescence movies
        suff = '_2P';
        ext = 'tif';
        ...
```

This would reduce code where alternative extensions are required:
``` 
 jsonPath = fullfile(fileparts(dat.expFilePath(expRef, 'parameters', 'master')),...
      [expRef, '_parameters.json']);
```
becomes `` jsonPath = dat.expFilePath(expRef, 'parameters', 'master', 'json');``

nsteinme, 2018-05-09:
Good idea. Let's also be moving toward a world where we're mostly querying alyx directly to find out directly which files exist and where...

k1o0, 2018-05-09:
Alyx.expFilePath returns existing records from Alyx, whereas dat.expFilePath simply constructs a path string for use in the saving of data.

k1o0, 2019-05-14:
This was done a while back 3b6a32270963c9dc92cc79858dbd3ab2f5d23e58.  In case you're interested Nick, I'll be working on better Alyx search integration over the next couple of weeks.  On the dev branch getSessions now works very well for searching sessions and converting expRefs into uuids.

closed 2019-05-14 by k1o0