cortex-lab/Rigbox #52: timelinehw.json broken?

nsteinme, 2018-05-09:
Just tried to load one and it didn't work - 

`q = loadjson('\\zubjects.cortexlab.net\Subjects\Beadle\2018-05-03\2\TimelineHW.json')
`

also looks wrong in a text editor. Not sure what's going on?

k1o0, 2018-05-09:
Okay, I'll test a fix now.

k1o0, 2018-05-09:
Fix'd:

```
p = '\\zubjects.cortexlab.net\Subjects\PC022\2018-05-09\1';
fid = fopen(fullfile(p, 'TimelineHW.json'), 'r');
jsonData = fscanf(fid, '%s');
fclose(fid);
hw = jsondecode(jsonData);

fid = fopen(fullfile(p, '2018-05-09_1_PC022_parameters.json'), 'r');
jsonData = fscanf(fid, '%s');
fclose(fid);
pars = jsondecode(jsonData);

fid = fopen(fullfile(p, '2018-05-09_1_PC022_hardwareInfo.json'), 'r');
jsonData = fscanf(fid, '%s');
fclose(fid);
hwInfo = jsondecode(jsonData);
```

NB: I wrote a function called obj2struct (called by obj2json) that cleanly converts pretty much any object to a MATLAB struct, so keep it in mind if you want to save anything else to JSON.

Re-open this issue if you encounter another problem.

closed 2018-05-09 by k1o0