cortex-lab/Rigbox #53: weight logger broken/needs update

nsteinme, 2018-05-10:
It used to be the case that selecting a mouse in the "experiments" panel of MC would also select it in the "log" panel. And, when a mouse is selected, the star no longer shows up to indicate the current weight reading. 

Either we can repair these bugs that have been introduced, or perhaps simply make a new button in the alyx panel. The button will say "[no scale]" and grayed out when no scale is connected, or will say "record x.xx g" when a weight reading is present. This will be a simpler and more convenient system, and the "log" tab can go away entirely.

k1o0, 2018-05-10:
The first point is not a bug.  We both actually decided to have the log and experiment tabs independent from one another as people often come in to take weights without wishing to change experiment.  

The log tab is very informative about historical mouse weight so we decided to encourage people to switch tabs so they could see the animal's weight with respect to the minima (people can't be arsed to click 'View Subject History').  Now the panel also changes colour when a mouse crosses a weight threshold so perhaps it's less important.  The log tab should still remain as being able to view comments from previous sessions (and previous activity in general) is very useful.  

Regarding the star, yes that is a bug.  The star still appears once the reading has been taken, and as it did before if the user is not logged in.  I couldn't figure out why.  There's also a very old bug where MC throws an error if you try to record a weight before the scale has updated.

nsteinme, 2018-05-10:
I'm confused about your analysis of user interaction. You say that people can't be asked to click View Subj History, which is the same number of clicks as clicking Log - but then you also want them to now do an extra two clicks to select the subject that they [usually] already selected in the other panel... 

I don't object to keeping the log tab for anyone who wants to use it. I think the button to log weight from alyx panel directly makes a lot of sense though...

k1o0, 2018-05-10:
The point is that before we added the Alyx Panel everyone used the log tab to weigh their mice, and people continued to do so.  When the training room is busy people want to weigh their mice without interrupting the user who is about to run another mouse.  We had a problem when a lot of mice were dropping below 80% weight and no one was noticing because the only way to view that graph was by weighing the mouse, then clicking view subject history.  At the time it made perfect sense to have the same plot in the log window which everyone was using anyway, especially as the stand-alone was introduced a little later and wasn't used so much at the start.

I agree that behaviour has probably changed now that people use the stand-alone panel and that the panel tells you if the mouse is below a weight threshold, so these features could change.  I'll look into the weigh button but it may not be feasible as multiple figures would have to share the same scale object.

k1o0, 2018-05-10:
Fix'd the star bug: 55e925d6fcd389e946a25a0541242144aa441946

k1o0, 2018-05-25:
6ebf03f4b1fd6a081a06066e7a0b6626b526ca13 Weight button now listens to the scales object.  You have to hit the button within 10s of the weight updating or it reverts back to entering the weight manually.  Try it out and re-open this issue if you have any problems.

closed 2018-05-25 by k1o0