cortex-lab/Rigbox #59: update warning

ZeitgeberH, 2018-06-19:
Hi,
I am using the lite-branch. The experiment runs, but in matlab console,  it gives me some strange warning/errors (see below) after i added a licking detector input. Could you provide some guidance on how to solve this issue?
Best,
Zeta

Warning: Error caught in signals updates: length of updates = 2, length newNUpdates = 14 
> In MyWorldExpPanel/expUpdate (line 138)
  In eui.ExpPanel>@(varargin)p.expUpdate(varargin{:}) (line 85)
  In srv.StimulusControl/onWSReceived (line 192)
  In srv.StimulusControl>@(varargin)obj.onWSReceived(varargin{:}) (line 177)

k1o0, 2018-10-09:
Hi, this warning can be ignored for now.  It's something we'll keep looking into but it only really affects the plotting of things in the experiment panel.

k1o0, 2018-10-25:
See  #72

k1o0, 2019-08-13:
Closing due to duplication

closed 2019-08-13 by k1o0