cortex-lab/Rigbox #61: server reconnection

HernandoMV, 2018-09-26:
Hi, I am having a problem reconnecting from Master to Stimulus only after Stimulus (srv.ExpServer) crashes. On 'mc' gui, if I click on Stimulus, I see 'Dissconnected from Stimulus' in the log.
This problem is only solved afte I restart the Master computer (just restarting Matlab doesn't solve it, neither restarting Stimulus computer). 
This piece of code reproduces the problem:
```
uri = 'ws://Stimulus2:2014';
socket = net.entropy_mill.websocket.Client(uri);
socket.connect()  % crashes here :(

onError: java.net.UnknownHostException: Stimulus2
```

Any ideas on how to sove it? Thanks,

HernandoMV, 2018-10-02:
We solved the problem by replacing the computer name with the ip address in the remote.mat file

k1o0, 2018-10-09:
I've updated the documentation to use the IP address instead of the computer name.  It'll be in the next version of the instructions

closed 2018-10-09 by k1o0