cortex-lab/Rigbox #62: Signals updates too slow

k1o0, 2018-10-09:
Since updating Rigbox to send inputs and parameters updates to mc, the panels have become far too slow due to the CPU requirements of deserializing so many structures and cell arrays.  This needs to be fixed urgently as it's becoming difficult to train multiple mice at once.   Perhaps signals should only send updates once every 100ms?  Or maybe there's a more effective way of sending the updates?

nsteinme, 2018-10-09:
I think making them once every 50 or 100ms or so should do it... I think it
is easy: in the main loop of signals, just put a "if toc(tstart)>0.05;
tstart=tic(); ... send updates ... end"

k1o0, 2018-10-24:
276cd8e989a502dd59c419773083cec01a498eed tested and works

closed 2018-10-24 by k1o0