cortex-lab/Rigbox #63: Online plotting in signals choiceWorld

k1o0, 2018-10-09:
The ExpPanelFun for choiceWorld fails to plot correctly as the updates arrive out of order.  This needs to be fixed to deal with this.

jaib1, 2019-03-16:
what's the status of this?

k1o0, 2019-03-18:
@laurax452 Since that update to choiceWorldExpPanel do the psychometric plots look accurate?

jaib1, 2019-03-18:
I think so, but I stopped training using tasks that plot, so I only saw this improvement for a few days. 
You could try asking Hamish..? I have a feeling he mentioned his weren't plotting correctly recently but not sure what tasks he was running..

laurax452, 2019-03-18:
That was me Laura by the way, didn't realise I had been signed out on this computer.

k1o0, 2019-10-03:
Since there's been no activity here I'm closing this issue.  Re-open if you experience problems again.

closed 2019-10-03 by k1o0