cortex-lab/Rigbox #71: Opaque error in hw.calibrate

k1o0, 2018-10-24:
'Assignment has more non-singleton rhs dimensions than non-singleton subscripts' error on line 56 when readGrams fails.  Should throw a more helpful error.

jaib1, 2018-12-12:
Related to this, we should do a better job of throwing our own created error messages throughout Rigbox with meaningful info for clients

k1o0, 2019-02-08:
Although there is still no good error handling, this error hasn't occurred since e4cdf247ac6354600eb319fcb17db4b957bb7196
The issue was that there was a leading char byte when the scale had a negative value (a '-').

jaib1, 2019-02-08:
This error occurred two days ago when calibrating. The issue was that if the scale is tared before calibrating, this error occurs

k1o0, 2019-02-08:
Can you please prioritize this?  I don't have a scale so you need to do the following:

- [ ] Investigate and fix the error when the scale is tared*
- [ ] Create a WeighingScale test object that includes the exact sort of data returned by an fscanf of the ES-300HA scale

*Taring is problematic because approach 0g, the scale automatically re-tares when there are very small weight changes, however this shouldn't be causing a read error

jaib1, 2019-02-12:
I tried recreating the issue, but didn't have any problems...will try again with Hamish tomorrow

k1o0, 2019-09-16:
Assuming @kevin-j-miller's fixes in v2.3 have solved this.  Will reopen if necessary.
https://github.com/cortex-lab/Rigbox/commit/4898beba0d28ca2f301fe3f976cfb3be22b78898#diff-1194d7b78aec701d3bbe281389fdfd95

closed 2019-09-16 by k1o0