cortex-lab/Rigbox #81: Submodules and dependencies

k1o0, 2018-12-07:
Currently we have a load of submodules in Rigbox, however they are gradually becoming more and more dependent of Rigbox functions.  We should be careful to ensure that these submodules can be used alone, otherwise we may as well fold them into Rigbox entirely...

jaib1, 2018-12-07:
Along these lines, I've been thinking about ways to do package/file re-organization so *Signals* can be completely standalone (e.g. moving some files from Rigbox +exp to *Signals* ?)

k1o0, 2018-12-07:
If you're thinking about SignalsExp then that won't solve the problem; there's absolutely no way of running SignalsExp without Rigbox.  inferParameters could be moved though, but only to the cortexlab signals fork as @dendritic probably doesn't want to support that function.