cortex-lab/Rigbox #82: Improving Code Readability

jaib1, 2018-12-07:
While working through code refactoring, I think it's important we focus on improving code readability at the same time. Two aspects of this jump out to me currently:

1) When writing new code, we should document thoroughly before pushing commits

2) Looking over old code, it would be helpful to 
    a) remove unnecessary or commented code that hinders readability
    b) if commented code is not to be deleted, an explanation explaining the purpose of having that code in, and why it's currently commented, should be added

k1o0, 2018-12-10:
I don't quite understand 2b.  First, I think we should just go ahead and delete code, or create an issue first, rather than having discussions in the actual code itself.  TODO and FIXME tags are useful as you can search for them but otherwise comments regarding suggestions, bugs and questions should be made into GitHub issues instead.  Second, do you mean having meta comments about the comments themselves?  Could you give an example?  

I agree that there should be lots more comments, although not every line needs justification, for instance in a while loop `i = i + 1` should be pretty clear, or these two exmples: `rig.connect()`, `rotateticklabel(obj.WeightAxes.Handle, 45)`.  However I think we should for sure err on the side of over explaining things.

jaib1, 2018-12-10:
> I agree that there should be lots more comments, although not every line needs justification, for instance in a while loop i = i + 1 should be pretty clear, or these two exmples: rig.connect(), [...]

Of course, I meant summary descriptions of things like methods and any code that you think may be unintuitive, so I think we're on the same page here. 

> Second, do you mean having meta comments about the comments themselves? Could you give an example?

Yes, for example: https://github.com/cortex-lab/Rigbox/blob/dev/cortexlab/%2Bgit/update.m#L31-L37

A comment above this could be: "This code tries to do <explanation>, but fails. <possible alternative>"

I think it's very helpful having these types of comments in the code itself if they can be kept relatively short, rather than searching through old issues just to see if something exists that references this.

k1o0, 2018-12-10:
Yeah code like that should just be deleted rather than commented out.  That's laziness on my part...

closed 2020-02-16 by k1o0