cortex-lab/Rigbox #89: hw.Timeline: make local directory regardless of what's on the server

petersaj, 2019-01-09:
Julie had a problem yesterday when doing an experiment with mpep and timeline - mpep creates an experiment folder on the server very quickly, and for some reason Timeline only makes a local experiment folder if there's no server experiment folder, so it breaks. Can you make this change:

hw.Timeline line 159
if ~dat.expExists(expRef); mkdir(fileparts(localPath)); end % if the folder doesn't exist, create it

to this:
if ~exist(fileparts(localPath),'dir'); mkdir(fileparts(localPath)); end % if the folder doesn't exist, create it

k1o0, 2019-01-09:
Hmm, that may not solve your problem as those two lines achive the same thing: they check if the experiment folder exists (e.g. '\\zubjects.cortexlab.net\Subjects\default\2019-01-08\1') and if not creates it.  What's the precise error you're getting?

petersaj, 2019-01-09:
it's because timeline's trying to write to a file in the local path, but it can't open that file because the local path doesn't exist, here's the whole block just for reference:
```matlab 
            if obj.WriteBufferToDisk
                fprintf(1, 'opening binary file for writing\n');
                localPath = dat.expFilePath(expRef, 'timeline', 'local'); % get the local exp data path
                if ~dat.expExists(expRef); mkdir(fileparts(localPath)); end % if the folder doesn't exist, create it
                obj.DataFID = fopen([localPath(1:end-4) '.dat'], 'w'); % open a binary data file
                % save params now so if things crash later you at least have this record of the data type and size so you can load the dat
                parfid = fopen([localPath(1:end-4) '.par'], 'w'); % open a parameter file
                fprintf(parfid, 'type = %s\n', obj.AquiredDataType); % record the data type
                fprintf(parfid, 'nChannels = %d\n', numInputChannels); % record the number of channels
                fprintf(parfid, 'Fs = %d\n', obj.DaqSampleRate); % record the DAQ sample date
                fclose(parfid); % close the file
            end
```
and it errors at this line:
```matlab
fprintf(parfid, 'type = %s\n', obj.AquiredDataType); % record the data type
````
because it's trying to write to parfid, but parfid isn't a valid because it should be [localPath(1:end-4) '.par'] and localPath doesn't exist

k1o0, 2019-01-09:
Ahh of course, sorry for being so slow haha

petersaj, 2019-01-09:
that's ok fast responses from Lithuania :beer:

k1o0, 2019-01-09:
Changed on dev  fcd6dcb0fc03cadcd65d29503a20c666e92c0ffc
By the way as of right now dev should be quite stable so it's worth moving your rig over to it.  If everything continues working I'll merge into master next week.

closed 2019-01-09 by k1o0