cortex-lab/Rigbox #93: Creating a Rigbox FAQ

jaib1, 2019-01-25:
I think a Rigbox FAQ would be really helpful, particularly with troubleshooting common errors (e.g. paths errors, hardware/daq errors, submodule errors (particularly in signals and alyx), PTB/other dependency errors)

k1o0, 2019-01-25:
There is already an FAQ in the instructions for this very purpose, however it should be expanded.  I had also promised Laura that I would add a list of steps on the Wiki for what to do if there's an error.

jaib1, 2019-01-26:
Right it should be expanded. We should also include the hardware and software installation instructions and FAQ directly in this repo.

k1o0, 2019-01-26:
Do you want to concentrate on this?  I'll send you the Google Doc, it would be good to revise and expand it, then potentially add it to the paper as an appendix.  Also we should get rig of rigbox-lite.

k1o0, 2019-10-03:
e7659901973661d123cb103f4204287b3986f7f4

closed 2020-02-16 by k1o0