cortex-lab/Rigbox #97: Treating dead mice by mc

mkrumin, 2019-02-01:
mc does not allow to run experiments with an animal, which is marked dead on Alyx. This is fine. However, even after 'resurrecting' an animal on Alyx (it was erroneously marked dead and reduced), mc does not allow to run an experiment with this mouse.

mkrumin, 2019-02-03:
Yes, Charu was the responsible user for this animal. But this doesn't mean we shouldn't be able to run experiments on it. Is this rule there to avoid listing all the stock animals?

k1o0, 2019-02-04:
Yes it's to avoid seeing the many, many stock mice we have.  At one point we had requested a change so that if the mouse was on protocol 3 it was no longer a stock animal (https://github.com/cortex-lab/alyx/issues/382) but that [isn't the case in the code](
https://github.com/cortex-lab/alyx/blob/e84e9c351cd89724d3da8bc2c5b1dea6b14ce0ba/alyx/subjects/views.py#L21-L25).

Double check that the mice is under protocol 3 and if it is I'll reopen that issue

mkrumin, 2019-02-04:
Right now the animal (CR010) is on protocol 3, and it should've been on protocol 3 during than experiment.  Can't test the mc behavior now with it, as it is already dead. We might find other animals on protocol 3 with Charu being a responsible user.