cortex-lab/Rigbox #125: Event times dropped 1200 seconds mid experiment

k1o0, 2019-03-19:
Event logs for one rig on dev branch went back by 1200 seconds in the middle of a session:

**Trial number vs log times (s)**
[image]

All event times should be strictly monotonically increasing.  May be related to issue #45.  Last related changes to the code were commits https://github.com/cortex-lab/Rigbox/commit/0e5e4004e1daecdfefc2486994016899249a1990 & https://github.com/cortex-lab/signals/commit/1c9127de66f3edc4c2b0063019b3e9045929a21c.

May have something to do with Timeline or the behaviour of GetSecs().

jaib1, 2019-03-19:
why does this have a signals label? this is urgent. the issue was a new task was accidentally started on a rig currently running a task, and instead of the usual "this rig is in use" message, the new task actually started and overrode the current task.

k1o0, 2019-03-19:
I added the signals label because it's specifically to do with the SignalsExp class and the Signals Registry files.  

>  instead of the usual "this rig is in use" message, the new task actually started and overrode the current task
In that case, it may be related to issue #6.

k1o0, 2019-03-19:
Consider the following:
https://github.com/cortex-lab/Rigbox/blob/master/%2Bsrv/StimulusControl.m#L86-L103
https://github.com/cortex-lab/Rigbox/blob/master/%2Beui/MControl.m#L456-L461

I imagine the problem here is a failure on the MControl side to properly grey-out the BeginExp button.  We need to work out under which circumstances this occurs.  I've also just pushed a hotfix to the dev branch (https://github.com/cortex-lab/Rigbox/commit/5d28fb18a986e56d565e89fd39056f24fe07269f) which should hopefully avoid this happening again in the future.  @jaib1 please test this asap on a rig.  You can temporarily comment out the bits of code in MControl that grey out the start button so that you can do this.  

By the way this is exactly the type of thing that would be caught by a set of tests haha.

jaib1, 2019-03-19:
> I added the signals label because it's specifically to do with the SignalsExp class and the Signals Registry files.

I'm confused - it seems that this issue could occur for any experiment type (e.g. ChoiceWorld overriding Signals), and not just Signals? If not, could you explain how you know this could only occur with SignalsExp?

> I imagine the problem here is a failure on the MControl side to properly grey-out the BeginExp button.

Based on Pip and Laura's description, I think the error occurs in 'eui.MControl/rigConnected' - that 'rig.ExpRunning' is somehow read as false even when an experiment is running (because the 'questdlg' never pops up). https://github.com/cortex-lab/Rigbox/blob/e0eca9edd6b293cbab96d54d1a6b701b74d83d94/%2Beui/MControl.m#L409

I wasn't able to naturally recreate this, but when I did so forcibly, your solution worked, but I thought it might be nice to stop the problem before it reaches the stim server (and before the MC GUI changes), so I also added a hotfix to 'eui.MControl/beginExp': https://github.com/cortex-lab/Rigbox/commit/c5ac87a6a09112d49175d9239b6a0ddae02fb1a2

k1o0, 2019-03-19:
> I'm confused - it seems that this issue could occur for any experiment type (e.g. ChoiceWorld overriding Signals), and not just Signals? If not, could you explain how you know this could only occur [...]

I didn't know this at the time.  It was you that informed me that the issue was a new task was accidentally started on a rig currently running a task in the comment above, which was posted after I had added the signals label.  

I'd prefer it if you kept the previous code and investigated why that dialog didn't pop up otherwise we're losing some functionality here.  The previous fix already ensures that a currently running experiment won't be affected.

k1o0, 2019-03-19:
The previous code uses the get.ExpRunnning method, which in turn uses the same downstream methods as get.Status.  If the issue is that expServer didn't send the correct response than this fix won't work.  The other possibility is that `if expRef` always evaluates to true.  In which case, maybe change to `if ~isempty(expRef)`

jaib1, 2019-03-19:
> I added the signals label because it's specifically to do with the SignalsExp class and the Signals Registry files.

You made this comment after I explained that the issue was overriding a currently running experiment =p but that's ok, I just wanted to make sure this isn't purely a *Signals* issue.

> I'd prefer it if you kept the previous code

can you be more specific about "previous code" ?

> The previous code uses the get.ExpRunnning method

is this "previous code" the same as above? again can you say what exactly this is? I didn't remove any code, only made a typo fix ("Runnning" to "Running") and added code (to 'beginExp').

> The other possibility is that if expRef always evaluates to true

this is definitely not the case - I investigated this.

>  If the issue is that expServer didn't send the correct response than this fix won't work

I assume you're talking about my fix to 'beginExp'. Yes, this is true, which is why I left your code in. That way, if/when the problem occurs again, we can have a better sense of where it is occurring, depending on if my or your error message stops execution.

> investigated why that dialog didn't pop up otherwise we're losing some functionality here

this is of course what I tried to do, and what is what I meant when I said "I couldn't recreate the problem naturally" (i.e. by trying different ways of running an experiment on a rig that already had an experiment running). if you have other ways of trying to investigate this I'm all ears. can you explain what you mean by "losing some functionality" ?

In general, can you try to be as verbose as possible - I think you often are, but in this case there's a lot of ambiguity, at least for me : ) I know being verbose can be unnecessary at times, but enough of the time it's very helpful for me

k1o0, 2019-03-19:
My mistake, I thought that dialog code was occurring in beginExp and had been replaced by your commit.  Also I edited my original comment to be clearer.

k1o0, 2019-03-19:
As far as I'm concerned this is now fixed as it shouldn't break anyone's experiments in the future.  Although we don't know why that 'run' message still got through I imagine we're not going to find a solution any time soon as it's an extremely rare occurrence, we can't replicate it and don't have any tests for this yet.  I suggest we re-open if we find two experiments running simultaneously again, or have more information as to why it happened in the first place.

If you'd prefer to leave it open then I suggest removing the priority tag and added the wontfix label.

closed 2019-03-20 by jaib1