cortex-lab/Rigbox #127: Add Rigbox config files

k1o0, 2019-03-19:
Related to #123.  The following configuration files would be helpful to users who wish to set up Rigbox from scratch: 

- [ ] configure Timeline
- [ ] configure Window
- [ ] configure sockets
- [ ] configure services
- [ ] configure paths
- [ ] configure hardware inputs and outputs
- [ ] configure audio devices
- [ ] configure Alyx (the client)

These could be placed in doc/setup/ and be developed on a new branch that's merged into dev each time a new config file is created.

Pros to this approach:
1. It helps users configure their objects the 'right' way using validation methods to add parameters e.g. srv.StimulusControl/create, hw.DaqRotaryEncoder/wiringInfo, hw.Timeline/wiringInfo, hw.Timeline/addInput
2. Demonstrates features that not many users know about, for instance adding timeline and path override parameters to their hardware files, how services can be configured for various applications.
3. Scripts are more useful than other types of documentation as they can request user input and help users setup thing whilst at the same time acting as a tutorial through demonstration.
4. We could do away with example binary files such as the hardware and remote MAT files, which aren't easy to track in GitHub.

jaib1, 2019-06-26:
should we also include docs on calibrations? e.g. how to perform gamma and reward valve calibrations?

k1o0, 2019-06-26:
These should be in readthedocs.io

k1o0, 2019-09-16:
Done in v2.3

closed 2019-09-16 by k1o0