cortex-lab/Rigbox #129: W crashes the srv.server in zym1 and 2

laurax452, 2019-03-20:
Using W to open the water valve in zym 1 and 2 causes a crash in the srv.server of those computers.
Spacebar works fine and reward delivery is working normally.
Can be a bit of a pain when flushing the lines through...
Error message said something about channel output issue and the DAQ controller....

k1o0, 2019-03-20:
This may have something to do with the recent changes made by @jaib1.  Mind investigating?

jaib1, 2019-03-20:
yup can investigate, but won't have time to until next week - asked Laura to assign it to you in case you have any immediate thoughts? Don't think the recent changes (just doing MATLAB/PTB updates) would affect this, particularly since the software was updated on all zym rigs, and zym 3 and 4 open the water valve using keyboard 'w' just fine?

k1o0, 2019-03-20:
Something must have changed about the hardware file.  At some point you made some changes to the reward output for Anna, could it be to do with that?

jaib1, 2019-03-20:
>  At some point you made some changes to the reward output for Anna, could it be to do with that

that was back pre new year's, and those were reverted. don't know how long this issue has been, since Pip said he doesn't typically use keyboard 'w', and Laura hasn't trained on the zym rigs since she's been back, until this week.

k1o0, 2019-03-20:
I see the hardware file for ZYM1 was modified yesterday (19th).  However modified it may have caused this error.  It's not easy for my to troubleshoot by my self remotely.  I would need to remote in when the tubes have been emptied and would need the TeamView login details (sent on Slack).

jaib1, 2019-03-20:
> I see the hardware file for ZYM1 was modified yesterday (19th)

This is when Pip and I were troubleshooting the audio device issue (when we found out ASIO drivers are no longer supported by PTB). We changed the hardware.mat files on all zym rigs, so again doubt this is the issue, but I suppose it could be possible that something affected the daqControllers only on zym1 and zym2 somehow. 

Anyway, I'll look into it next week, it's not terribly urgent.

jaib1, 2019-03-25:
was a daqController issue - wrong signal generator for the reward valve.

closed 2019-03-25 by jaib1