cortex-lab/Rigbox #143: Assigning a vector as a parameter to a visual stimulus

jaib1, 2019-04-19:
```matlab rect = vis.patch(t, 'rectangle');
rect.colour = p.color;
rect.dims = p.size;

try
  p.color = [1 1 1]';
  p.Size = [20 10]';
catch
end
```

Both the assignments to ```rect.colour``` and ```rect.dims``` fail with the error:

> Error using horzcat
> Dimensions of arrays being concatenated are not consistent.
> 
> Error in sig.transfer.mapn (line 27)
>   val = f(inpvals{:});
> 
> Error in sig.node.OriginSignal/post (line 15)
>       affectedIdxs = submit(this.Node.NetId, this.Node.Id, v);

jaib1, 2019-04-22:
Related to this, I think the way parameters are used in Signals in general can be improved in a few small ways. One way is in discriminating global from conditional parameters in an exp def. Instead of the current use of having conditional parameters set as row vectors, how about something more explicit, like having sub structs (one for global parameters, another for conditional parameters) within the 'parameters' exp def input arg? @k1o0

k1o0, 2019-04-29:
Moved discussion to new issue.  In terms of this 'bug', the issue is simply one of transposing the parameter before using it, e.g. `stimulus.colour = p.colour'` or `stimulus.dims= p.dims.map(@transpose)`.  

This problem occurs because some visual element fields are required to be row vectors and conditional parameers are always column vectors:
https://github.com/cortex-lab/signals/blob/a38950eb55f48feac1df040b3f1b666572ec7b89/%2Bvis/grating.m#L46-L47

Perhaps a good solution to this would be to transpose the array within the visual element code:
```
newelem.colour = iff(size(newelem.colour,1) > size(newelem.colour,2), ...
newelem.colour', newelem.colour); % Ensure row vector
gratingLayer.minColour = l.*[newelem.colour 0];
```

or if the following is syntax is implemented in Signals:
```
gratingLayer.minColour = l.*[newelem.colour(:); 0];
```

k1o0, 2019-04-29:
I closed the above pull requests because the base made things messy.  See the below and let me know what you think:

Changes would be: `gratingLayer.minColour = l.*[newelem.colour 0];` -> `gratingLayer.minColour = l.*[newelem.colour(:); 0].';`

> Two things to consider before merging:
>
>1. transpose could be removed if the renderer is fine with column vectors
>2. perhaps this change should be rejected altogether: for optimal performance the transpose step could be left up to the user, e.g. `stim.colour = p.colour.'`.

jaib1, 2019-05-02:
ok, so should I go ahead and make this change in 'vis.patch', 'vis.grating', 'vis.image', 'vis.grid'? And add it to the [Tutorials for pape](https://github.com/cortex-lab/signals/pull/10#pullrequestreview-232992506)r pull request? Anything else I'm missing?

k1o0, 2019-05-02:
Double check that all fields containing vectors can be assigned either shape, i.e. `dims` and `sigma`.  Otherwise it may be confusing for users if `stim.dims = [50; 50]` doesn't work but `stim.colour = [1; 1; 1]` does.