cortex-lab/Rigbox #144: Improving definition of global and conditional parameters

k1o0, 2019-04-29:
Related to this, I think the way parameters are used in Signals in general can be improved in a few small ways. One way is in discriminating global from conditional parameters in an exp def. Instead of the current use of having conditional parameters set as row vectors, how about something more explicit, like having sub structs (one for global parameters, another for conditional parameters) within the 'parameters' exp def input arg? @k1o0

_Originally posted by @jaib1 in https://github.com/cortex-lab/Rigbox/issues/143#issuecomment-485448750_

k1o0, 2019-04-29:
I think any approach needs explaining to the user in the documentation.  It's certainly a good idea to explain this in the try-catch block of all example definition functions.  Currently an unequal trial conditions error is caught by the Parameters class which is usually first encountered when loading parameters in mc.  We could make this error message clearer or even check for this in inferParams (although that's run just before they're loaded into Parameters anyway):

https://github.com/cortex-lab/Rigbox/blob/9521b66bf8d3bc4e37f703b141213d097dbd9777/%2Bexp/Parameters.m#L173-L174
https://github.com/cortex-lab/Rigbox/blob/9521b66bf8d3bc4e37f703b141213d097dbd9777/%2Bexp/inferParameters.m#L34-L36

I think once the user is aware that columns correspond to different trial conditions there's not much more to it.  Additionally the global and conditional parameters could be separated in the example functions to make the difference more obvious to new users:

```

%% Parameter defaults
% See timeSampler for full details on what values the *Delay paramters can
% take.  Conditional perameters are defined as having ncols > 1, where each
% column is a condition.  All conditional parameters must have the same
% number of columns.
try
%%% CONDITIONAL PARAMETERS
% Contrast detection set
c = [1 0.5 0.25 0.12 0.06 0];
C = [c, zeros(1, numel(c)-1); zeros(1, numel(c)-1), c];
p.stimulusContrast = C;
p.repeatIncorrect = abs(diff(C,1)) > 0.25; % | all(C==0);

%%% GLOBAL PARAMETERS
p.onsetToneFrequency = 5000;
p.interactiveDelay = 0.4;
p.onsetToneAmplitude = 0.15;
p.responseWindow = Inf;
p.stimulusAzimuth = 90;
p.noiseBurstAmp = 0.01;
p.noiseBurstDur = 0.5;
p.rewardSize = 3;
p.rewardKey = 'r';
p.stimulusOrientation = [0, 0]';
p.spatialFrequency = 0.19; % Prusky & Douglas, 2004
p.interTrialDelay = 0.5;
p.wheelGain = 5;
p.preStimulusDelay = [0 0.1 0.09]';
catch % ex
%    disp(getReport(ex, 'extended', 'hyperlinks', 'on'))
end
```

Would there be a way to implement your suggested improvement without making the code more bulky? It would be a little cumbersome having to write the following:

```
parameters.conditional.contrast = [1 0.5 0.25 0.12 0.06 0];
parameters.global.colour = [0.5 0.5 0.5];
```

jaib1, 2019-04-30:
I agree that any approach needs better documentation.

Regarding this:

> Would there be a way to implement your suggested improvement without making the code more bulky? It would be a little cumbersome having to write the following:

I was thinking that it could be done in a similar approach as to how individual stimuli are assigned to the 'VisStim' handler in an exp def. Going off your code snippet in the above comment, could just do something like:

```matlab
c.contrast = [1 0.5 0.25 0.12 0.06 0];
g.colour = [0.5 0.5 0.5];
parameters.conditional  = c;
parameters.global = g;
```

I'm not married to this approach however - I'll think more about this.