cortex-lab/Rigbox #153: ACW Exp Panel error

jaib1, 2019-05-14:
Warning: Error occurred while executing the listener callback for event Refresh defined for class
eui.MControl:
Error using imrotate
Expected input number 2, ANGLE, to be a scalar.

Error in imrotate>parse_inputs (line 227)
validateattributes(ang,{'numeric'},{'real','scalar'},mfilename,'ANGLE',2);

Error in imrotate (line 67)
[A,ang,method,bbox] = parse_inputs(args{:});

Error in screenImage (line 31)
gratL =
imrotate(repmat(sin([1:gratSize]/gratSize*2*pi*gratSize/pixPerDeg*sf),gratSize,1).*gw*cL,ori,'bilinear','crop');

Error in advancedChoiceWorldExpPanel/processUpdates (line 271)
                            [x,y,im] = screenImage(p);

Error in advancedChoiceWorldExpPanel/update (line 46)
            processUpdates(obj); % update labels with latest signal values

Error in eui.MControl>@(~,~)panel.update() (line 587)
            event.listener(obj, 'Refresh', @(~,~)panel.update())];

Error in eui.MControl>@(~,~)notify(obj,'Refresh')

Error in timer/timercb (line 30)
    		feval(val{1}, obj, eventStruct, val{2:end});

Error in timercb (line 13)
timercb(t, varargin{2:end});

Error in bui.Axes>@(varargin)obj.processMouseMovement(varargin{:}) (line 67)
        event.listener(figureEvents, 'MouseMotion', @obj.processMouseMovement)

Error in bui.GraphicsObjectEvents/callMotionListeners (line 43)
      notify(obj, 'MouseMotion', bui.MouseEvent(src));

Error in bui.GraphicsObjectEvents>@(varargin)obj.callMotionListeners(varargin{:}) (line 26)
      set(graphicsHandle, 'WindowButtonMotionFcn', @obj.callMotionListeners); 
> In eui.MControl>@(~,~)notify(obj,'Refresh')
  In timer/timercb (line 30)
  In timercb (line 13)
  In bui.Axes>@(varargin)obj.processMouseMovement(varargin{:}) (line 67)
  In bui.GraphicsObjectEvents/callMotionListeners (line 43)
  In bui.GraphicsObjectEvents>@(varargin)obj.callMotionListeners(varargin{:}) (line 26)

k1o0, 2019-05-28:
The error message that you pasted says that `ori` is not scalar.  Have you made any changes to the orientation parameter?  Look at line 8 of screenImage.  My version still works as normal.

k1o0, 2019-09-16:
No activity on this.  Reopen if/when required.

closed 2019-09-16 by k1o0