cortex-lab/Rigbox #158: 'map' calls a function which doesn't have any outputs

jaib1, 2019-05-16:
e.g.

```
net = sig.Net;
os = net.origin('os');
m = os.map(@(x) disp(x));
os.post(1);
```
> Too many output arguments.
> 
> Error in sig.transfer.map (line 9)
>   val = f(wv);
> 
> Error in sig.node.OriginSignal/post (line 15)
>       affectedIdxs = submit(this.Node.NetId, this.Node.Id, v);

k1o0, 2019-05-18:
I wouldn't classify this as a bug.  What value would `m` take if the mapped function had no outputs?  My understanding is that `map` cannot possibly work without output assignment: what would you be mapping?

jaib1, 2019-05-18:
Hm yes you're right, I guess I wanted `map` to map the "action" of a function that doesn't have a specific output var, but it doesn't really make sense to do this

closed 2019-05-18 by jaib1