cortex-lab/Rigbox #165: Chrono Signal bleeds into analog inputs

jaib1, 2019-06-06:
In a saved `hardware.mat` file, when the elements in `timeline.Inputs.name` have a disordered match with the elements in `timeline.UseInputs` , the chrono signal bleeds into the analog inputs not connected to the reward valve

k1o0, 2019-06-07:
This is very surprising.  Could you share a couple of example timeline files?

jaib1, 2019-06-07:
if you look at some 'default' sessions from yesterday you should see them. Kevin has the data too

k1o0, 2019-06-07:
Hmm I looked at numbers 30 to 33 and the order in which the channels were added is identical.  The channels to use is determined with intersect, which returns an ordered array: 
https://github.com/cortex-lab/Rigbox/blob/c98a2477e37fd68a62f3a98988b93d6ba1e61640/%2Bhw/Timeline.m#L513

In none of the above sessions is the order of UseInputs and Inputs the same in the hardware file.  Do you have a specific session number where they were?

I am fairly sure that the order of 'UseInputs' relative to the Inputs structure is inconsequential.  I did however notice that you renamed the inputs (c.f. '2019-06-06_32_default' and '2019-06-06_31_default').  This _does_ change the order in which the channels are added.

jaib1, 2019-06-07:
`In none of the above sessions is the order of UseInputs and Inputs the same in the hardware file.`

I don't know how you can tell this from the saved data - where did you access the `UseInputs` properties in the saved data? I can assure you we used a `hardware.mat` file in which the order of the `UseInputs` and `Inputs` properties had an ordered match, in which chrono did bleed through, and another instance where those two properties had a disordered match, in which chrono did not bleed through, multiple times. Which exact sessions these were, I'm not sure - Kevin may have kept track, and he may have data (e.g. saved figures) showing this.

jaib1, 2019-06-07:
e.g.  
`load('\\zubjects\subjects\default\2019-06-06\20\2019-06-06_20_default_Timeline.mat')`
 `figure, plot(Timeline.rawDAQData(100:end-100,2))`

here is an example showing chrono bleeding through on the piezo.

jaib1, 2019-06-07:
The ordered vs disordered match was just a correlate we noticed - I think I agree with this: 

```I am fairly sure that the order of `UseInputs` relative to the Inputs structure is inconsequential.```

I guess the question here is, do you know how changing the `Inputs` or `UseInputs` property would cause the phenomenon above?

k1o0, 2019-06-07:
> I don't know how you can tell this from the saved data

You can view the state of the hardware file when the experiment was run by loading the `_hardwareInfo.json` file.  

> e.g.
load('\\zubjects\subjects\default\2019-06-06\20\2019-06-06_20_default_Timeline.mat')
figure, plot(Timeline.rawDAQData(100:end-100,2))
> 
> here is an example showing chrono bleeding through on the piezo.

Thanks I'll take a look.  Is there a session number under the exact conditions, but with no bleed-through and a 'disordered' useInputs list?

> I guess the question here is, do you know how changing the Inputs or UseInputs property would cause the phenomenon above?

No I don't.  The order shouldn't matter because they are added in alphabetical order regardless.  What could conceivably matter is the input name, as that does affect the order in which the channels are added, initialized and started (in theory this shouldn't matter though).

k1o0, 2019-06-07:
I looked at session number 20 and the order of useInputs and Inputs is the same:

```
hw = loadjson(dat.expFilePath('2019-06-06_20_default', 'hw-info', 'master', 'json'));
inputsList = cellfun(@(a)a.name, hw.timeline.Inputs, 'uni',0);
useInputsList = cellstr(hw.timeline.UseInputs);
isequal(inputsList, useInputsList)
```

jaib1, 2019-06-07:
```Thanks I'll take a look. Is there a session number under the exact conditions, but with no bleed-through and a 'disordered' useInputs list?```

Not that I'm aware of - we noticed the bleed-through only when `UseInputs` had an ordered match with `Inputs` (as in that session above).

mkrumin, 2019-06-07:
@k1o0 are you sure the channels are added in alphabetical order? If this is the case then this is not how it is supposed to be. We need to be able to explicitly define the order in which the channels are sampled on the NI board, it is critical (and not by naming them so that they are in alphabetical order).
For example, we have a weak piezo signal, which is affected by other channels. For this signal to be reliably recorded we a) measure it differentially b) have a dummy (short-circuited zero) signal before and after, to avoid any bleed-through from A/D. We have this type of setup on ZCAMP3 (B-Scope), and I think the timeline on kilotrode rig has the same.
I think you did set it up that way (the order of adding matters), and we cannot change it to just be alphabetical.

We can probably avoid bleedthrough from a strong chrono signal if we add a dummy signal immediately    after it (or just sample another strong signal after the chrono, which is unlikely to be affected by it). 
Strong signal means signal from a device able to provide enough current (not like bare piezo chip).

jaib1, 2019-06-07:
`load('\\zubjects\subjects\default\2019-06-06\20\2019-06-06_33_default_Timeline.mat')`
`figure, plot(Timeline33.rawDAQData(100:end-100,4))`

here is a session in which there is a disordered match between 'UseInputs' and 'Inputs', and there is no chrono bleedthrough on the piezo signal.

k1o0, 2019-06-07:
@mkrumin The channels and measurement types are defined by the Inputs structure.  They shouldn't be affected by the order in which they're added to the DAQ session object.  For example adding a single ended ai0 before a differential ai2 should be the same as adding those channels in the reverse order (i.e. adding a differential ai2 before a single ended ai0).

@jaib1 
> Not that I'm aware of - we noticed the bleed-through only when UseInputs had an ordered match with Inputs (as in that session above).

If you don't have any sessions where the order was mismatched why did you conclude that the order is what's causing this?  

I seems to me that people are confusing the order of the inputs list with the input configurations (that is their associated channels and measurement types).  An input attached to ai0 doesn't have to be first in the Inputs struct just because it's the first channel on the board.

jaib1, 2019-06-07:
`An input attached to ai0 doesn't have to be first in the Inputs struct just because it's the first channel on the board.`

Yes, I get this.

`If you don't have any sessions where the order was mismatched why did you conclude that the order is what's causing this?`

Initially we were trying to solve the chrono bleedthrough issue on the piezo analog input. In trying to solve the bleedthrough, we realized that when you have a disordered match between `UseInputs` and `Inputs`, the bleedthrough does not occur, and when you have an ordered match, it does occur.

k1o0, 2019-06-07:
on ZREDTHREE edit the timeline object in the following way:
```
timeline.UseInputs = circshift(timeline.UseInputs ,1); % don't forget to save!
timeline.UseInputs % check the list is the same as below
  ans = 
    1×4 cell array
    ["kevin"]     ["chrono"]    ["rewardvalve"]    ["piezoanalog2"]

```
Make sure the names of the inputs are the same.  The Inputs struct should have the same names but with 'kevin' last.  Run another default experiment using 'pavlovianWorld' with Timeline enabled and check that there is no longer bleed-through.

jaib1, 2019-06-07:
Yes, I would expect that this wouldn't cause chrono bleedthrough. Is there a reason you would want me to test this anyway?

k1o0, 2019-06-07:
> Yes, I would expect that this wouldn't cause chrono bleedthrough. Is there a reason you would want me to test this anyway?

Because your assumption hasn't been tested and I think it's wrong.  The order of UseInputs doesn't affect the order in which the channels are added.  I think it is more likely that the order in which the channels are added to the main session by Timeline would affect things.  By doing this test we will see if when the order in which the channels are added is the same, the order of the UseInputs list still causes bleed-through. This seems impossible to me but we can only know be testing it.

mkrumin, 2019-06-07:
> The order of UseInputs doesn't affect the order in which the channels are added. 

Then how do we control the order of the channels? I though this was the way to do that.

kevin-j-miller, 2019-06-07:
>
> I am fairly sure that the order of `UseInputs` relative to the Inputs
> structure is inconsequential.

This might be true! What we know is that the order in UseInputs itself is
consequential, because changing this (from ['chrono', 'piezo1', 'piezo2',
'rewardvalve'] to ['chrono', 'rewardvalve', 'piezo2',' piezo1']) caused the
bleedthrough to go away. It might be important that both piezos were set up
as single-ended, and that piezo2 was shorted to the analog-input ground.
This probably isn't the best configuration, but it's the one we were using.

Hypothesis: The order in UseInputs is the order in which the channels are
sampled (as Krumin suggests). Sampling a weak channel right after a strong
channel causes bleedthrough into the weak channel, because the ground
hasn't completely cleared. This hypothesis predicts, among other things,
that the bleedthrough should be of opposite polarity to the signal it's
bleeding from.

One of the experiments Jai and I did yesterday bears on the alphabetical
hypothesis (though it's probably not conclusive).  We renamed "rewardvalve"
to "arewardvalve", which caused no change -- there was still bleedthrough
from 'chrono' into 'piezo1'. Maybe the perfect experiment would be renaming
it to "crewardvalve"?

k1o0, 2019-06-07:
Currently there isn't a way to control the order, as they're added alphabetically.  Theoretically the order in which the channels are added to the DAQ session shouldn't affect anything right? Do you know if this is addressed anywhere in the official documentation?

I can make a change so that if the array column index is not empty, it uses this to determine the order, however if the order does affect things and it is not documented we should report it to MATLAB as a bug.

mkrumin, 2019-06-07:
I just checked an example Timeline file from `ZCAMP3` (`'\\zubjects.cortexlab.net\Subjects\JL037\2019-06-06\1318\2019-06-06_1318_JL037_Timeline.mat`')
The order of channels in `timeline.UseInputs` in the relevant `hardware.mat`  is the same as in the actual data file, and it is not alphabetical. And you can see there how we put the piezo channel between the two dummies to avoid (reduce) bleedthrough:

This is from the hardware.mat
```
>> timeline.UseInputs'

ans =

  14×1 cell array

    {'chrono'           }
    {'syncEcho'         }
    {'photoDiode'       }
    {'audioMonitor'     }
    {'piezoCommand'     }
    {'piezoPosition'    }
    {'laserPower'       }
    {'pockelsControl'   }
    {'neuralFrames'     }
    {'rotaryEncoder'    }
    {'rewardCommand'    }
    {'dummy1'           }
    {'piezoLickDetector'}
    {'dummy2'           }
```

This is from the actual recorded Timeline structure
```
>> {Timeline.hw.inputs.name}'

ans =

  14×1 cell array

    {'chrono'           }
    {'syncEcho'         }
    {'photoDiode'       }
    {'audioMonitor'     }
    {'piezoCommand'     }
    {'piezoPosition'    }
    {'laserPower'       }
    {'pockelsControl'   }
    {'neuralFrames'     }
    {'rotaryEncoder'    }
    {'rewardCommand'    }
    {'dummy1'           }
    {'piezoLickDetector'}
    {'dummy2'           }

```

mkrumin, 2019-06-07:
> Theoretically the order in which the channels are added to the DAQ session shouldn't affect anything right? Do you know if this is addressed anywhere in the official documentation?

We use multiplexing boards (mot simultaneously sampling boards), and as a consequence get 'ghosting' 
[Here](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019KzzSAE&l=en-GB) is a page on NI website explaining how to avoid ghosting.
Measuring unamplified signals differentially is a different thing - it is about collecting less noise on the way to DAQ, not about ghosting.

All this was first figured out/implemented by @nsteinme , but I cannot find the original references right now

k1o0, 2019-06-07:
Ah yes you're quite right, it does add them in the order of UseInputs:
https://github.com/cortex-lab/Rigbox/blob/c98a2477e37fd68a62f3a98988b93d6ba1e61640/%2Bhw/Timeline.m#L513
https://github.com/cortex-lab/Rigbox/blob/c98a2477e37fd68a62f3a98988b93d6ba1e61640/%2Bhw/Timeline.m#L520-L522

I was under the impression that ghosting was the result of channels being physically adjacent to one another and not the order in which they're added to the main session.  Is this not so?

k1o0, 2019-06-07:
Okay, in light of all this, it seems this issue is not really relevant to Rigbox as it's a device problem.  This info should be put in the lab wiki for prosperity and perhaps briefly mentioned in the Timeline documentation (under the `init` method?)

mkrumin, 2019-06-07:
> I was under the impression that ghosting was the result of channels being physically adjacent to one another and not the order in which they're added to the main session. Is this not so?

In multiplexing boards there is a single A/D (why we have a cap on _total_ sampling rate), which is sampling the channels in a cycle. It has certain time constants, so a strong signal on one can affect the measurement on the next one. On that page I linked above techniques to avoid ghosting are discussed, and one of them is 'sample less frequently'. So, I believe our ghosting problem is more prominent on systems with many input channels (effectively higher overall sampling rate, smaller dt). But this is solvable by introducing a shorted dummy channel (or we can play around with the exact order of channels), if you have capacity on the board, which is usually the case for us.

k1o0, 2019-06-07:
It sounds as though it samples the channels in the order they were added to the session then.  For the above reasons I'll close this issue.  Feel free to open again if you think the Rigbox code needs changing.

jaib1, 2019-06-07:
```Okay, in light of all this, it seems this issue is not really relevant to Rigbox as it's a device problem. This info should be put in the lab wiki for prosperity and perhaps briefly mentioned in the Timeline documentation (under the init method?)```

Do you want to do this and then close (i.e. specifically mention when ghosting may occur, and how to deal with it) ? Can I assign this to you, since you were so sure that `UseInputs` didn't affect the order in which the inputs are used : )

mkrumin, 2019-06-07:
In addition, I think channels with names starting with 'dummy' are not getting npy-ed. I don't know where to check, but if this is true, this also needs to go into the documentation near ghosting.

nsteinme, 2019-06-07:
Btw the solution Michael says (grounding channels that are adjacent to the
problematic one, in order of sampling) is exactly the solution we used at
my rig, indeed. However, the 'correct' solution is to build a small
amplifier so that the problematic signals are not corruptible in this way.
I.e. the lick detector produces a voltage but no current and is therefore
susceptible to interference via ghosting - if it was amplified (even 1x
gain) so there was a current source behind it, then you wouldn't have to do
the dummy channel thing. I just couldn't figure out how to build a small
amplifier easily enough (Miles may remember that we tried this for a while
and got stuck).

kevin-j-miller, 2019-06-07:
Thanks, Nick & Michael! I'll try and reproduce that setup in the
behavior-room rigs that I'm working with.
I'm hoping to move soon to optical lick detectors for the two-port setup,
but if those turn out to have issues I'll keep the amplifier trick in mind
for optimizing the piezos. I guess this is the same principle as a
zero-gain headstage amplifier for ephys -- make the signal ghost-proof as
close as possible to the place where it's generated.

nsteinme, 2019-06-07:
That's how I understand it, yes - but given that I failed to build such an
EE101-level amplifier perhaps you shouldn't rely much on my understanding
here ;)

jaib1, 2019-06-07:
On voltage buffer amplifiers (voltage followers/unity gain buffers)

http://www.ni.com/product-documentation/4494/en/

k1o0, 2019-06-10:
Updated wiki

closed 2019-06-10 by k1o0