cortex-lab/Rigbox #170: Reading and writing over multiple repositories

k1o0, 2019-06-14:
Currently the user can define only one path for each repository.

**Problem**: When migrating to a new repository or splitting experiments over multiple repositories causes issues for analysis and loading previous parameters.

**Solution**: Allow each repo field to be an array, perhaps with a flag to indicate whether the repos distribute data (i.e. during migrations) or are fallbacks (i.e. during server failures #155).

```
p.mainRepository = ["\\server1\Data"; "\\server2\Data"];
```

We already have local vs master built in to the dat package, perhaps that should be expanded to include other options:

```
>> numel(dat.expFilePath(ref, 'parameters'))

ans = 

   3

>> filePath = dat.expFilePath(ref, 'parameters', 'fallback')

filePath = 

   \\server2\Data
```

Things to consider:
* Searching multiple repositories may be slower as it will requires access to multiple locations then perhaps a collation step.  Affect of performance would be minimal though.  
* How would the package deal with duplicates and conflicting data?  How does the user prioritize the repositories?

k1o0, 2019-06-14:
Assigning you two for your opinions on the matter

mkrumin, 2019-06-14:
Not a coherent description, just a few points about how I think things should be implemented

- `dat.paths` should have `mainRepository`, and in addition a list of all other historical repositories, which might contain data. I think `mainRepository` should still return only one result, for compatibility.
- `dat.listExps` should check in the `mainRepository`, and also in all the historical ones
- `dat.expPath` should return `mainRepository` for a new experiment, or, if an experiment with that `ExpRef` already exists (what is the criterion for this? folder with some data in?), return the location of that experiment (as this will be mostly used in the analysis for loading data), maybe with an additional flag that the location is historical, which might be read-only?

What other functions in `+dat` package will be affected?

We will almost definitely end up with some experiments split up into multiple locations, e.g. during analysis somebody's code will write results into `mainRepository`, while the raw data was in `oldRepository`

nsteinme, 2019-06-14:
I think Michael's suggestions seem sensible to me. But are we (ha: you) not
ready to go to alyx for this? Why not just ask alyx where the most recent
experiment is (and, why not ask alyx where new ones should go)?

k1o0, 2019-06-17:
Yeah `getSessions` now works very well and `getFile` works on https://github.com/cortex-lab/alyx-matlab/pull/46.  I think that's a good solution for loading the previous parameters when logged in on mc.  I think people still use the dat package in non-Rigbox experiments where they don't yet use Alyx, right @mkrumin?

mkrumin, 2019-06-17:
I agree with Nick that eventually we want to query Alyx about everything, including weather, but I don't think we are ready yet. 

- One reason is that not all files are getting registered on Alyx (e.g. we have 2p data, ultrasound data, and maybe some other files, which are copied manually to the server post experiment, or just that the acquisition code is not using Alyx at all - mpep-run experiments?). 
- (We should avoid this manual copying thing, and have scripts instead that copy files to the server while registering them on Alyx. I should probably write these scripts.)
- Also, during the analysis, I think almost nobody (or actually nobody) is querying Alyx, but uses Rigbox/+dat package to figure out the location of the files (and I believe some people just hard-code the paths, but we are not going to solve this type of asocial behavior).
- In addition, can we move files (file paths) on Alyx? As an example I had run a couple of experiments with the wrong animal name, and then, after the experiment I just manually moved and renamed the files to their intended location on the server, so that the analysis scripts (which use dat package) would work properly. 

As an immediate solution we need the dat package to check multiple paths, so that at least for people who use it things will be transparent. We can also try and implement Alyx querying under the hood of the dat package on a separate branch, and see if it works well?

closed 2019-07-15 by k1o0