cortex-lab/Rigbox #173: Make zserver.cortexlab.net/Data the mainRepository

mkrumin, 2019-06-17:
On startup of mc (and expServer?) we have this automatic update, which rolls back any changes to +dat/paths.m
zserver.cortexlab.net/Data is our mainRepository for some time now (and will probably never be zubject again), so we should change this in the +dat/paths.m 'permanently'

k1o0, 2019-06-17:
The file in Rigbox won't always be the cortexlab one, especially as more people start to use Rigbox.  Users should be putting their custom paths file in their MATLAB directory and ensuring that it is at the top of the paths list.  This is already mentioned in the documentation.  Another solution is to save your paths in the rig's config file:

```
clear all
name = 'ZRIG'; % example rig name
repoLoc = 'C:\NewPath'; % example path
p = dat.paths(name);

if exist(fullfile(p.rigConfig,'paths.mat'), 'file') == 2
  load(fullfile(p.rigConfig, 'paths'), 'paths');
end

paths.mainRepository = repoLoc;
save(fullfile(p.rigConfig, 'paths'), 'paths')
```

jaib1, 2019-06-17:
```This is already mentioned in the documentation```

Where is this mentioned? If you're talking about this:

https://github.com/cortex-lab/Rigbox/blob/master/docs/setup/paths_config.m

do you want to update this file?

k1o0, 2019-06-17:
Yes I'll update that and the readme.  I haven't worked out readthedocs yet so I'll let you deal with that ;)

mkrumin, 2019-06-18:
Can we just change that one line in paths.m so that on the next code update (which is currently still forced) the things will not roll back to zubjects? This is a **non-temporary lab-wide** change we are making, not for specific rigs. Also, for non-cortexlab users I think it doesn't matter what [wrong] line is there, as they will need to override it anyway. I don't think it is feasible at the moment to ask all the rigs to manually define the paths (which are really the same for almost everyone).

k1o0, 2019-06-19:
I made the change in https://github.com/cortex-lab/Rigbox/pull/174/commits/9183daf24b12c0ba94a56ad363e023911c8678db (part of the PR).

k1o0, 2019-07-15:
zserver now main in the paths template

closed 2019-07-15 by k1o0