cortex-lab/Rigbox #175: Avoiding paths confusion

k1o0, 2019-06-19:
Related to #173 and #137, how do we reduce confusion among users over the paths file?  In cortexlab people have repeatedly had the same problem for years now: namely they inadvertently set the MATLAB paths incorrectly and end up calling the wrong paths file.  Cortexlab users are inexplicably against copying the +dat/paths.m file in their MATLAB directory, and this still causes problems if their current directory is Rigbox.  

Suggestion 1:
Remove the paths file from the repo entirely and have the users create it themselves.  Although there could be a GUI function for picking the paths I think this is inflexible and actually more awkward than simply having users paste their paths into an m file.  We could have both options available.  The drawback is that the paths file is a nice example file and without a GUI script you can't validate the function and ensure hostname specific custom paths are loaded.  

Suggestion 2:
Have the paths file in the docs directory outside of the dat package.  It can then be pointed to as an example and addRigboxPaths could even copy it to the user's MATLAB directory for them (asking them permission first).

jaib1, 2019-06-24:
I am in favor of removing `dat.paths` completely, and having an example file in the `docs` folder that tells the user how to set-up `dat.paths` after install #173

k1o0, 2019-06-25:
- [ ] Update the docs so that during setup user must copy and edit template paths
- [ ] Delete +dat/paths.m file and create docs/paths_template.m
- [ ] Add warning if no paths file exists during addRigboxPaths
- [ ] Merge into master

closed 2019-07-24 by k1o0