cortex-lab/Rigbox #190: Changing saved remote path for a subject which has already run in the same day

jaib1, 2019-08-01:
e.g.
```
Failed because experiment ref '2019-08-01_2_test does not exist
```

k1o0, 2019-08-01:
Mind elaborating for the slower ones among us?

jaib1, 2019-08-01:
Related, this fails https://github.com/cortex-lab/alyx-matlab/blob/ed0718231205b1dd3a050f9d9c56addc198de813/%40Alyx/newExp.m#L64-L65

for 'default' and 'test' mouse if, e.g., they are being used by two different people on the same day on different rigs

k1o0, 2019-08-01:
From what I understand this was a local paths issue.  Re-open if necessary.

jaib1, 2019-08-02:
> Related, this fails https://github.com/cortex-lab/alyx-matlab/blob/ed0718231205b1dd3a050f9d9c56addc198de813/%40Alyx/newExp.m#L64-L65

for 'default' and 'test' mouse if, e.g., they are being used on two different rigs on the same day

^ This is still an issue, though not a huge deal. Any ideas on a fix?

k1o0, 2019-08-02:
There are often multiple experiments run under test and default each day with no problem, so I have a string suspicion that this is a paths issue.  Either way you'll have to provide a lot more information for me to troubleshoot this.

k1o0, 2019-08-08:
Following up on this.  Please close if you can't reproduce the error.

jaib1, 2019-08-08:
The error in `newExp` still occurs as I described. The current fix involves either manually copying or manually deleting expRef folders from the remote server to the local rig. any long-term fix is a bit problematic because of the unique cases of `default` and `test`. Just adding an exception for these cases might be the simplest fix (e.g. if the `expRef` contains `default` or `test`, it doesn't matter if an expRef folder already exists for the given expRef). However, even this isn't ideal, in the unlikely but possible situation that someone decides to run an animal on two separate rigs on the same day.

k1o0, 2019-08-12:
> ahh, I wasn't clearing the subject's folder on MC. Because zrig1 and zrig3 save to different remotes, there were 5 total folders I had to clear, and I only cleared 4...
> There's also the issue of two different rigs saving to two different remotes, in which case you get this error if you try to run the same subject on a second rig in the same day

At any one time there must be one and only one main, master repository.  The mc computer must save parameters to the same repository as is set by the stimulus computer that will run said experiment.  The issue here was not that the two stim computers had different remotes, but that the mc and one of the stims had different remotes.  

Experiments are iterated based on what exists in all remotes, so you won't get these errors when changing repos so long as you do it correctly, i.e. by making the old repo an alternate and setting the new repo for both mc and the stim computer.

closed 2019-08-12 by k1o0

jaib1, 2019-08-12:
```
i.e. by making the old repo an alternate and setting the new repo for both mc and the stim computer.
```
To be clear, all "old" remote repos will have to be set as alternate repos in `dat.paths` for all the SCs and MCs that are used to run a particular subject who is saved on multiple remote repos, correct? 

E.g. if a subject  is run on zrig1 and zrig2, and zrig1 saves to `zubjects/Subjects`, and zrig2 saves to `zserver/Data/Subjects`, zrig1 should have `zserver/Data/Subjects` as an alternate repo, zrig2 should have `zubjects/Subjects` as an alternate repo, and the mc computer can have either remote as its main repo, as long as the other remote is an alternate repo. Is this right?

k1o0, 2019-08-13:
No.  The rigs should all have the same main repo location. The alternates are there for searching old experiments.  Think of them as archives.  

Here are the steps:
- When you begin an experiment mc looks for the most recent experiment reference by looking everywhere, including the alternate locations.  
- It creates a new unique experiment reference.
- It saves a set of parameters for that experiment locally and in the main repo location _only_.  We don't want any more redundancy here, that's what server backups are for.
- It tells the chosen stimulus rig to start a new experiment.
- The stimulus computer looks for the parameters in the main repo location _only_.  Again, the alternate locations are for finding old experiments, not new ones.
- At the end of the experiment the stimulus computer saves the data locally and in the main repo location _only_. 

So if you want to run experiments both the mc and stim computers should have the same main repo location.  In cortexlab we want all our current experiments to be saved on zserver/Data/Subjects.  When this changes to the new server you should change the paths for _all rigs_ so that the main repo location is this new server.  In order to find old experiments, the other locations may be set as alternate repos, i.e.
```
mainRepository = '\\znew\Subjects';
main2Repository = '\\zserver\Data\Subjects';
main3Repository = '\\zubjects\Subjects';
```

This will allow users to load old experiment files using the dat package and more importantly means that if you change the main repo location in the middle of the day, mc won't mistakenly create a duplicate experiment reference in the main repo because it will be able to see all the previous experiment references for a given subject.

In typical Jai style I know you're about to respond with 'how about we change this so that users can save to multiple different main repos on different rigs?', and in typical Miles style I'm going to preempt this by saying, no, we won't do that because it would be a lot of work and no one needs or has ever needed that feature.  If a time comes when we need to implement a more complicated and flexible file repository system, we'll consider the options then.  Such a system would probably use Alyx, which comes with its own caveats.

jaib1, 2019-08-13:
> In typical Jai style I know you're about to respond with 'how about we change this so that users can save to multiple different main repos on different rigs?'

ha, close but not quite. Rather, why not have SC1 and SC2, controlled by the same MC, be able to save to different remotes? I can easily imagine a situation where a lab would want this. Also, tangentially related, is there a reason for saving local exp data saved on mc, if that experiment info will be saved on sc and the remote?

k1o0, 2019-08-13:
> why not have SC1 and SC2, controlled by the same MC, be able to save to different remotes 
See above.

> is there a reason for MC to save experiment info files locally, given that the same files or saved on SC and remotely anyway?
MC only saves the parameters.  For both mc and the stimulus computer, saving locally first means that the data are saved somewhere before an attempt is made to save to the remote repo.  Therefore if the server is inaccessible and an error is thrown, the data won't be lost.