cortex-lab/Rigbox #204: Running timeline can be closed by other rig accidentally

petersaj, 2019-10-14:
During an experiment my timeline shut itself down with this message (I was running AP047 on the rig, someone else was running AP049 on a different rig):

[14-10-2019 14:56:22] mpep: 'ExpStart AP049 20191014 1' from 128.40.198.199:1103
Timeline already running, stopping first
Timeline for '2019-10-14_1_AP047' stopped and saved successfully.

It seems like instead of stopping the current experiment and replacing it with the new one, Timeline should refuse the connection and the hijacking computer should get an error message?

k1o0, 2019-10-17:
Interesting.  Strange that this has literally never happened before... Anyway it's a easy fix but will take some time as I need to create tests.

petersaj, 2019-10-17:
It ended up being because that training rig computer had temporarily been used as the imaging rig computer when it broke, so the findServices pointed to the imaging timeline computer and the remote.mat file had timeline as an available. So that was just a mistake on our part, but seems like a safe fix would be having timeline refuse new experiments while it's already running one?

k1o0, 2019-10-17:
I thought as much.  I won't change the behaviour of Timeline but I'll change the behaviour of tl.MpepListeners so that if timeline is running it will not re-start.

closed 2020-02-16 by k1o0