cortex-lab/Rigbox #215: [MC] every other experiment results in error "inconsistent experiment ref from rig"

synapticlee, 2019-11-11:
Hi, since a few days, I have a weird bug where every other experiment I try to run results in a warning error "inconsistent experiment ref from rig" (srv.expServer does not crash though). As in, e.g. the odd numbered experiments might be fine and the even numbered experiments might crash.

k1o0, 2019-11-11:
I'll take a look.

k1o0, 2019-11-11:
In case @jaib1 or someone else wants to look into it at the rig, here are the relevant bits of code:

Below in MControl the expRef is created then passed to the ExpPanel constructor:
https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Beui/MControl.m#L577-L584

The experiment ref is then sent to expServer via the StimulusControl object (called `rig` here):
https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Beui/MControl.m#L589

https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Bsrv/StimulusControl.m#L127-L135

expServer should get the expRef from the io.WSJCommunicator object then immediately send it back with the 'starting' message:
https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Bsrv/expServer.m#L209-L227

The ref is then passed to the panel via notification of the 'ExpStarted' event:
https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Bsrv/StimulusControl.m#L210-L213

https://github.com/cortex-lab/Rigbox/blob/79f7db70af618e5b388b80942bf51566591caa9e/%2Beui/ExpPanel.m#L190-L196

I recently made a change to expServer (https://github.com/cortex-lab/Rigbox/commit/4fa283af3572dcedf659a5a5df3411b051e0c076) so the problem most likely stems from there.

jaib1, 2019-11-21:
@k100 did you look into this?

jaib1, 2019-12-29:
closing as this was most likely due to a paths issue. feel free to re-open if this occurs again

closed 2019-12-29 by jaib1