cortex-lab/Rigbox #216: typo/bug: gamma correction doesn't work

petersaj, 2019-11-15:
srv.expServer Line 368: turns off plotting the figure

hw.ptb.Window Line 712: uses the variable "upCrossings" which is only generated by line 688 if the figure plot is turned on

so as it's written gamma correction can't happen - this issue is fixed by generating the up/downcrossings outside of the figure-related code

k1o0, 2019-11-15:
Cheers for the info! Mind making a PR?

petersaj, 2019-11-15:
I've never done a pull request before - I thought that might happen if I attempted to push to master but it ended up just committing the code. Not sure if I should just leave it?

k1o0, 2019-11-15:
Pushed already by Andy (7d6b6010d9166632ba5a7181ed9414ea866480f7)

closed 2019-11-15 by k1o0