cortex-lab/Rigbox #259: storing of string lists is incorrect

anwarnunez, 2020-02-10:
If a value of a variable that gets saved under the `events.` is a string, then the output is simply concatenated resulting in something like:

"thisthatthisthatthisthatthisthat"

And it should instead be:
this,
that,
this,
that,
this,
that,
this,,
that,

Thanks!
Anwar

jaib1, 2020-02-10:
yes, basically there should be a new string in each element of an array whose length is the number of times the corresponding signal is updated, rather than one long concatenated string

k1o0, 2020-02-10:
Mind giving me some more context?  Signals handles basic data types in the same way as native MATLAB.  It looks to me that your signal is actually a char and not a string:

```
events = sig.Registry();
events.str = sig.test.create;
events.char = events.str.map(@char);
s = string(strsplit(which('addRigboxPaths'), filesep));
for i = 1:length(s), events.str.post(s(i)); end

data = logs(events)
data = 

  struct with fields:

     strValues: ["C:"    "Users"    "User"    "Documents"    "Github"    "rigbox"    "addRigboxPaths.m"]
      strTimes: [2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05]
    charValues: 'C:UsersUserDocumentsGithubrigboxaddRigboxPaths.m'
     charTimes: [2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05 2.0837e+05]
```

anwarnunez, 2020-02-10:
I have something like : `myvar = iff(var, 'this', 'that')` and `myvar` is
stored as `events.myvar = myvar`.

I don't know matlab well enough to know when it maps values to chars or
strings. however, this current behavior seems undesirable.

thanks!
anwar

k1o0, 2020-02-10:
In MATLAB you can define a string by using double quotes, and a char array with single quotes, so the above example could be changed to `myvar = iff(var, "this", "that")`

More info here: https://uk.mathworks.com/help/matlab/characters-and-strings.html

It's worth noting that events are always horizontally concatenated, so if you have a row vector and want to store its history, transpose it when assigning to the events registry and you'll get a matrix out.

closed 2020-02-10 by k1o0

anwarnunez, 2020-02-10:
ha! :facepalm:
sorry for the noise

k1o0, 2020-02-10:
No worries! All comments welcome.  The wider question here is whether to 'improve' the way Signals handles data over native MATLAB with things like automatic typecasting but I think this would probably lead to more bugs and confusion among users...