cortex-lab/Rigbox #266: differentiating Rigbox and submodule docs

jaib1, 2020-02-16:
some docs that are currently in Rigbox/docs should be moved to their appropriate submodules? 
e.g. `SignalsPrimer.html` and `AlyxMatlabPrimer.html` and their associated image files should be moved to the `signals` submodule and `alyx-matlab` submodule , respectively.

k1o0, 2020-02-17:
In terms of the links it's clearer and more simple to have them in a single location, that way the link paths in the scripts are short.  The scripts themselves are still in submodules.  At the end of the day we can just direct people to the Rigbox docs and ideally the HTML would be hosted on a Webserver anyway.

jaib1, 2020-02-17:
>  that way the link paths in the scripts are short.
ok sure, that makes sense for the Rigbox documentation. I guess another question is do we want to have html files as docs in the submodules? this could be redundant, but if the submodules are truly standalone then maybe they should also have their own html pages?