cortex-lab/Rigbox #278: No audio in eui.SignalsTest

dbirman, 2020-02-26:
**Describe the bug**
I'm don't get any audio when running experiment definition files through the new `eui.SignalsTest()`. This is using experiment definition files that ran previously in the `exp.ExpTest` and had audio that worked (although using the audio devices often caused matlab to crash unexpectedly). Running them through the mc + exp server generates audio normally. 

**To Reproduce**
For my setup I don't get sound with any of the experiment definition files that I have.

**Desktop (please complete the following information):**
Rigbox ver 2.5.0
MATLAB ver 2019b

**Additional context**
My installation appears to fail a lot of tests, so it's possible those are related to the above, but they don't appear related at first glance.

```
Failure Summary:

     Name                                                                                            Failed  Incomplete  Reason(s)
    =============================================================================================================================================
     AlyxPanel_test[BaseURL=https___test_alyx_internationalbrainlab_org]/test_viewSubjectHistory       X                 Failed by verification.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     AlyxPanel_test[BaseURL=https___test_alyx_internationalbrainlab_org]/test_giveWater                X                 Failed by verification.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     AlyxPanel_test[BaseURL=https___testDev_alyx_internationalbrainlab_org]/test_viewSubjectHistory    X                 Failed by verification.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     AlyxPanel_test[BaseURL=https___testDev_alyx_internationalbrainlab_org]/test_giveWater             X                 Failed by verification.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     dat_test/test_loadBlock                                                                           X         X       Errored.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     dat_test/test_expParams                                                                           X         X       Failed by assertion.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     dat_test/test_saveParamProfile                                                                    X         X       Errored.
    ---------------------------------------------------------------------------------------------------------------------------------------------
     Signals_test/test_size                                                                            X                 Failed by verification.

```

Thanks in advance for any help!

k1o0, 2020-02-26:
Audio devices seem to vary quite a bit across computers.  eui.SignalsTest tries to find a reasonable default.  It basically chooses the output device with the lowest reported latency:
https://github.com/cortex-lab/Rigbox/blob/616e8238373f6edc12316974f70c30ec1c9b344b/%2Beui/SignalsTest.m#L197-L202

If you provide more info about your computer's devices we could perhaps add more criteria for selecting the default device.  I'm wondering if you have a soundcard with multiple outputs like surround sound, digital and stereo.  If so then maybe we could ensure that it always selects a 2 channel output as default.  You can inspect the audio devices by running the first two lines of the above code.

You can also provide your own rig object as the second argument, e.g.
```
rig = hw.devices();
e = eui.SignalsTest('advancedChoiceWorld', rig);
```

or if you just want to change which is the default audio device, you can edit the Hardware.audioDevices struct:
```
e = eui.SignalsTest('advancedChoiceWorld');
e.Hardware.audioDevices

ans = 

  struct with fields:

          DeviceIndex: 4
       HostAudioAPIId: 13
     HostAudioAPIName: 'Windows WASAPI'
           DeviceName: 'default'
      NrInputChannels: 0
     NrOutputChannels: 2
      LowInputLatency: 0
     HighInputLatency: 0
     LowOutputLatency: 0.0030
    HighOutputLatency: 0.0100
    DefaultSampleRate: 48000
```

dbirman, 2020-02-26:
Thanks for the help, I was able to fix this for my machine by picking a specific audio device rather than the default. 

I was using a speaker plugged into a screen or the headphone port, but the criteria in SignalsTest was picking the "digital audio" channel, which I believe is the motherboard output? Seems like a reasonable default for 99% of cases (see below, the screen was device #7, the headphone port was device #9, the default pick by SignalsTest was #8). Channel #9, the headphone port, shares the lowest output latency, but in this setup it would never get picked because it shows up second in the list. 

A solution for making testing easier would be to add a button in SignalsTest which opens a panel where you can change which device is set to be the default audio channel? This way a user doesn't need to set up a hardware.mat file ahead of time, e.g. in this case I was just trying to plug in a speaker to quickly test it on my desktop before setting it up in a rig. To make things simple it could default to only showing the WASAPI devices when they are available, following the suggestion from the psychtoolbox api (http://psychtoolbox.org/docs/PsychPortAudio-GetDevices). 

Here's the output from calling `PsychPortAudio(‘GetDevices’)` on my computer:

```
{d.DeviceName}

ans =

  1×13 cell array

  Columns 1 through 6

    {'Microsoft Sound…'}    {'Headphones (2- …'}    {'PHL 276E8V (NVI…'}    {'LG Ultra HD (NV…'}    {'Digital Audio (…'}    {'PHL 276E8V (NVI…'}

  Columns 7 through 12

    {'LG Ultra HD (NV…'}    {'Digital Audio (…'}    {'Headphones (2- …'}    {'Output (NVIDIA …'}    {'Output ()'}    {'SPDIF Out (HD A…'}

  Column 13

    {'Headphones (HD …'}
```

```
{d.HostAudioAPIName}

ans =

  1×13 cell array

  Columns 1 through 9

    {'MME'}    {'MME'}    {'MME'}    {'MME'}    {'MME'}    {'Windows WASAPI'}    {'Windows WASAPI'}    {'Windows WASAPI'}    {'Windows WASAPI'}

  Columns 10 through 13

    {'Windows WDM-KS'}    {'Windows WDM-KS'}    {'Windows WDM-KS'}    {'Windows WDM-KS'}
```

```
[d.LowOutputLatency]

ans =

    0.0900    0.0900    0.0900    0.0900    0.0900    0.0030    0.0030    0.0027    0.0027    0.0100    0.0100    0.0100    0.0100
```

k1o0, 2020-02-28:
A drop-down menu option is a good idea.  There is currently a function called `hw.findDevice` that will play noise through all of your audio devices.  Pressing space bar returns the current device struct which you can then assign to your Hardware.audioDevices field.  Because the function will be renamed in the next version I will attach it here (be sure to rename it from .txt to .m)

[testAudioOutputDevices.txt](https://github.com/cortex-lab/Rigbox/files/4266300/testAudioOutputDevices.txt)

closed 2020-02-28 by k1o0

dbirman, 2020-02-28:
Great, that function is very helpful. Thank you!