cortex-lab/Rigbox #307: Installation issue

moraogando, 2020-10-12:
**Describe the bug**
A clear and concise description of what the bug is.  Feel free to paste in the error that was printed to the MATLAB command window.  

Hi there, this toolboox seems super cool! I'm having some issues trying to set it up though. It's possible that there's nothing wrong with the code but in stead I'm missing something, as I'm not a programmer. I wonder if you could help me out! I already installed Rigbox in my future SC, but when I try to do the same for the master computer I have problems. I installed all the dependencies and I didn't get warnings when I did addRigboxPaths().

I type "eui.SignalsTest' and the gui opens as expected, but I get the following error upon trying to start pong experiment:

>> eui.SignalsTest
PTB-INFO: Using modified PortAudio V19.6.0-devel, revision unknown

ans = 

  SignalsTest with properties:

          Hardware: [1×1 struct]
               Ref: '2020-10-11_1_test'
    LoggingDisplay: [1×1 UIControl]
          LivePlot: off
      ShowExpPanel: on
      SingleScreen: off
        Experiment: [0×0 exp.test.Signals]
       ParamEditor: [1×1 eui.ParamEditor]
       LivePlotFig: [0×0 Figure]
          ExpPanel: [0×0 eui.ExpPanel]
         IsRunning: 0

Warning: No gamma calibration available
Compiling all shaders matching C:\Users\UserName\Documents\MATLAB\Github\Rigbox\signals\+vis\slimshady * into a GLSL program.
Building a fragment shader:Reading shader from file C:\Users\UserName\Documents\MATLAB\Github\Rigbox\signals\+vis\slimshady.frag ...
Building a vertex shader:Reading shader from file C:\Users\UserName\Documents\MATLAB\Github\Rigbox\signals\+vis\slimshady.vert ...
ExpPanel destructor called
Warning: The following error was caught while executing 'eui.ExpPanel' class destructor:
Dot indexing is not supported for variables of this type.

Error in eui.ExpPanel/delete (line 199)
      if obj.Root.isvalid

Error in eui.ExpPanel.live (line 149)
            p = eui.SignalsExpPanel(parent, ref, params, logEntry);

Error in eui.SignalsTest/init (line 547)
        obj.ExpPanel = eui.ExpPanel.live(obj.ExpPanelBox, obj.Ref, obj.DummyRemote, p, 0);

Error in eui.SignalsTest/startStopExp (line 352)
        obj.init

Error in eui.SignalsTest>@(~,~)obj.startStopExp(obj.Ref) (line 461)
        'Callback', @(~,~)obj.startStopExp(obj.Ref)); 
> In eui/ExpPanel/live (line 149)
In eui/SignalsTest/init (line 547)
In eui/SignalsTest/startStopExp (line 352)
In eui.SignalsTest>@(~,~)obj.startStopExp(obj.Ref) (line 461) 
Error using uiextras.BoxPanel
Cannot define property 'Enable' in class 'BoxPanel' because the property has already been defined in the superclass 'Panel'.

Error in eui.ExpPanel/build (line 525)
      obj.Root = uiextras.BoxPanel('Parent', parent,...

Error in eui.ExpPanel (line 188)
      obj.build(parent);

Error in eui.SignalsExpPanel (line 40)
      obj = obj@eui.ExpPanel(parent, ref, params, logEntry);

Error in eui.ExpPanel.live (line 149)
            p = eui.SignalsExpPanel(parent, ref, params, logEntry);

Error in eui.SignalsTest/init (line 547)
        obj.ExpPanel = eui.ExpPanel.live(obj.ExpPanelBox, obj.Ref, obj.DummyRemote, p, 0);

Error in eui.SignalsTest/startStopExp (line 352)
        obj.init

Error in eui.SignalsTest>@(~,~)obj.startStopExp(obj.Ref) (line 461)
        'Callback', @(~,~)obj.startStopExp(obj.Ref));
 
Error while evaluating UIControl Callback.

**To Reproduce**
Steps to reproduce the behavior.  Be as specific as possible.  Does the error always occur?  Are there similar steps that don't produce the error, or is it specific to a particular experiment?

Every time unfortunately

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If GUI related, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Code version (e.g. `2.3.1`) - you can find this in the `CHANGELOG.md` file, or provide the Git commit info by pasting the output of running `git log -1` in Git Bash (`git.runCmd('git log -1')` from MATLAB)

version 2.4.0
(https://github.com/cortex-lab/Rigbox/releases/tag/2.4.0)

 - MATLAB version - run `ver` in the MATLAB command prompt and paste the output here.

MATLAB 2020b

moraogando, 2020-10-12:
I think it's a Matlab version issue, I switched to MATLAB 2020**a** and now it works just fine!

k1o0, 2020-11-11:
Great, thanks for letting me know.  I'll look into adding support for MATLAB 2020b in the next release.  If you encounter other issues consider moving to the v2.6 branch.

k1o0, 2021-01-05:
Just to let you know I've discovered an issue (#312) with the new version of Rigbox (2.6.0) on MATLAB 2020. I'm working on a fix but for now please make sure you don't update if you're on 2020a.

moraogando, 2021-01-10:
Thanks for the update! Will do